CarlFK/pici #13: Add pip + pyserial + tqdm + numpy

asinghani, 2024-01-30:
The UART on the FPGA should be passed through to the Pi, would be extremely handy to be able to use Python scripts to talk to it. Having some general convenience packages + pip itself would also be nice for making life easier.

CarlFK, 2024-01-30:
'''
+      - python3-full
+      - python3-pip
+      - python3-tqdm
+      - python3-numpy
'''
easy, done.

> UART on the FPGA should be passed through to the Pi
Is there something here that needs to happen? 

would this help? 
python3-serial

asinghani, 2024-01-30:
> UART on the FPGA should be passed through to the Pi

I believe it already is, but have not gotten a chance to test it yet. Will report back if there's an issue with it

> python3-serial
100% that would be suepr useful

closed 2024-05-27 by CarlFK