CarlFK/pici #16: Write a getting-started guide

asinghani, 2024-01-30:
It would be super nice to have a tutorial that can walk users through:

1. Setting up F4PGA and generating a bitstream
2. Copying it to the Pi and blinking an LED
3. Modify the design to use a UART (maybe have it be a loopback but it increments the character by 1)
4. Talk to the design over UART using screen and/or pyserial
5. Modify the design to use GPIOs (take two 4-bit numbers from GPIOs and add them, output to a third set of 5 GPIOs?)
6. Use RPi.GPIO library to interface with the GPIOs and pass data back and forth to the FPGA

CarlFK, 2024-03-05:
https://github.com/CarlFK/pici/wiki/Getting-Started
Does 1 and 2.  

3,4,5 are beyond getting started.  I think this is close enough:
LiteX - tio /dev/ttyUSB1  ... that shows the pi side of how to talk to it, but no fpga details.  

for #6: wire 2 fpga pins together, then use python gpio to verify the wire.

Abhijeettri141, 2024-08-14:
Is there anyone currently assigned to this feature? I am eager to contribute.

CarlFK, 2024-08-14:
for #6: wire 2 fpga pins together, then use python gpio to verify the wire.
done!
https://github.com/CarlFK/pici/tree/main/ansible/roles/ci/files/t1

CarlFK, 2024-08-14:
> Is there anyone currently assigned to this feature? I am eager to contribute.

I did a bunch of work, I think it is good enough, at least for now.  The main problem is it sends the user over to F4PGA examples documentation which is great docs until the part where those docs assume there is a local fpga, and then the user has to stop there can come back here.    I'm not really sure how much of a problem this really is.  Improving this seems like a lot of work for a little improvement. 

I think more stuff, even it is a bit crusty, will be better.  It will give people more to do, they will learn more, and in doing this we (you) will likely find things to improve the platform (what is per-installed on the pi.)