CarlFK/pici #19: linux_litex demo stuck on tftp sometimes

CarlFK, 2024-09-06:
following the steps on:

https://github.com/CarlFK/pici/wiki/Getting-Started#linux-litex

Sometimes it works, sometimes it gets stuck at:

Fetching from: UDP/6069
(no ascii spinner)

mithro, 2024-09-07:
How do I reproduce this?

mithro, 2024-09-07:
Carl says;
 * SSH into one of the machines using the "Linux cheatsheet" -- The random string that appears as you login is the password for the Pi.
 * cd `~/tests/linux_litex_t1`
 * `./t1.sh`
 * (If you get `[01:05:14.664] Error: Device file is locked by another process` then someone else is using it.) -- You should get a terminal with the standard Arty bitstream which looks like;
```
********************************************************
********************************************************
**        Avnet/Digilent Arty Evaluation Board        **
**        LEDs and switches GPIO Demonstration        **
********************************************************
********************************************************
**
Choose Task:
BTN0: Print PWM value.
BTN1: 'Cylon' LED display.
BTN2: Scrolling LED display.
BTN3: Return to this menu.

```
 * `ctrl-t q` to exit
 * The script will now try and load litex Linux stuff.

mithro, 2024-09-07:
```expect
#!/usr/bin/expect

spawn lsusb
expect "Future Technology Devices International"
expect eof
spawn openFPGALoader -b arty --reset --verbose
expect "found 1 devices"
expect eof
wait
spawn tio /dev/ttyUSB1
expect "Avnet/Digilent Arty Evaluation Board"
expect "LEDs and switches GPIO Demonstration"
expect "BTN3: Return to this menu."
close
spawn openFPGALoader -b arty top.bit --verbose
expect "found 1 devices"
expect eof
wait
set timeout 30
spawn tio /dev/ttyUSB1
expect "Memtest OK"
expect "Fetching from: UDP/6069"
expect "Downloaded 4652188 bytes from Image over TFTP to 0xc0000000"
expect "Executing booted program at 0x50000000"
expect "Liftoff!"
expect "*** VexRiscv BIOS ***"
#expect "*** Supervisor ***"
expect "No DTB passed to the kernel"
#expect "Welcome to Buildroot"
expect "ogin:"
send -- "root\r"
expect "root@buildroot"

expect "#"
# ip addr show
send -- "ip addr show\r"
expect "eth0"
expect "#"

# ip addr add 192.168.100.50/24 dev eth0
send -- "ip addr add "
sleep 1
send -- "192.168."
sleep 1
send -- "100.50/24 "
sleep 1
send -- "dev eth0\r"

expect "#"
# ip link set eth0 up
send -- "ip link set "
sleep 1
send -- "eth0 up\r"

expect "#"
# ip addr show
send -- "ip addr show\r"

expect "#"
# ping 192.168.100.100
send -- "ping -c 5 "
sleep 1
send -- "192.168.100.100\r"
expect "#"

interact
wait
```

CarlFK, 2026-11-13:
```
pi@pi3:~/Demos/linux_litex $ ./run_demo.sh 
+ cp -v tftp/emulator.bin tftp/Image tftp/rootfs.cpio tftp/rv32.dtb /srv/tftp
'tftp/emulator.bin' -> '/srv/tftp/emulator.bin'
'tftp/Image' -> '/srv/tftp/Image'
'tftp/rootfs.cpio' -> '/srv/tftp/rootfs.cpio'
'tftp/rv32.dtb' -> '/srv/tftp/rv32.dtb'
+ openFPGALoader -b arty top.bit
empty
Jtag frequency : requested 10.00MHz  -> real 10.00MHz 
Open file DONE
Parse file DONE
load program
Load SRAM: [==================================================] 100.00%
Done
Shift IR 35
ir: 1 isc_done 1 isc_ena 0 init 1 done 1
+ tio /dev/ttyUSB1
[06:10:23.266] tio 3.9
[06:10:23.268] Press ctrl-t q to quit
[06:10:23.269] Connected to /dev/ttyUSB1
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |00000000000000000000011111111111| delays: 26+-05
m0, b1: |00000000000000000000000000000000| delays: -
m0, b2: |00000000000000000000000000000000| delays: -
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b0 delays: 26+-05
m1, b0: |00000000000000000000011111111111| delays: 26+-05
m1, b1: |00000000000000000000000000000000| delays: -
m1, b2: |00000000000000000000000000000000| delays: -
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b0 delays: 26+-05
SDRAM now under hardware control
Memtest OK

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
Loading emulator.bin from flash...
Error: Invalid image length 0xffffffff
Booting from flash...
Error: Invalid image length 0xffffffff
Booting from network...
Local IP : 192.168.100.50
Remote IP: 192.168.100.100
Fetching from: UDP/6069
Unable to download Image over TFTP
Network boot failed
Unable to download Linux images, falling back to boot.bin
Unable to download boot.bin over TFTP
Network boot failed
No boot medium found

--============= Console ================--
litex> 
```